adafruit/Adafruit_MPU6050 #7: Calling .begin() multiple times has a memory leak

cimba007, 2021-03-06:
Multiple calls to begin and subsequently to _init will lead to the code below having a memory leak.

```
  delay(100);
  
  temp_sensor = new Adafruit_MPU6050_Temp(this);
  accel_sensor = new Adafruit_MPU6050_Accelerometer(this);
  gyro_sensor = new Adafruit_MPU6050_Gyro(this);
```

fix:

```
  delay(100);
  delete temp_sensor;
  delete accel_sensor;
  delete gyro_sensor;
  
  temp_sensor = new Adafruit_MPU6050_Temp(this);
  accel_sensor = new Adafruit_MPU6050_Accelerometer(this);
  gyro_sensor = new Adafruit_MPU6050_Gyro(this);
```

Proof: First Number is heap available and other numbers are acceleration X,Y,Z

```
00:56:51.542 -> 1199,0.29,0.23,8.37
00:56:53.938 -> 1172,0.26,0.24,8.25
00:56:56.334 -> 1145,0.21,0.27,8.25
00:56:58.760 -> 1118,0.17,0.31,8.18
00:57:01.150 -> 1091,0.26,0.25,8.26
00:57:03.540 -> 1064,0.26,0.33,8.41
00:57:05.946 -> 1037,0.22,0.21,8.01
00:57:08.381 -> 1010,0.28,0.31,8.30
00:57:10.780 -> 983,0.25,0.25,8.31
00:57:13.173 -> 956,0.26,0.29,8.26
```

Result .. on low memory ECU hangs or resets.

ladyada, 2021-03-07:
kk wanna submit a PR?

cimba007, 2021-03-07:
Sure .. it is my first PR so tell me if there is anything I can impove. Nice library btw. ;-)

caternuson, 2021-09-22:
Fixed with #8

closed 2021-09-22 by caternuson