adafruit/Adafruit_MPU6050 #11: Failed to find MPU6050 chip

ZCB1208, 2021-09-22:
the new version MPU6050 has the different DEVICE_ID

Change the Macro in “Adafruit_MPU6050.h” File:
   #define MPU6050_DEVICE_ID 0x68 -> #define MPU6050_DEVICE_ID 0x98 
can solve the problem .

I don't think this is the best solution to solve it. so I just put it in Issues.

I try the solution (add code to main.ino):

    #ifdefine MPU6050_DEVICE_ID 
    #undefine MPU6050_DEVICE_ID 
    #define MPU6050_DEVICE_ID 0x98
    #endif

but it did not work.

caternuson, 2021-09-22:
Where did you source the MPU6050's with `0x98` device IDs?

ZCB1208, 2021-09-23:
> Where did you source the MPU6050's with `0x98` device IDs?

From “https://www.taobao.com”  .  I come from China .

caternuson, 2021-09-23:
It's probably not a MPU6050.

closed 2021-09-23 by caternuson