adafruit/Adafruit_MPU6050 #18: Alternative to resetting clearing MPU on init and begin calls.

mahboud, 2022-03-07:
- Arduino board:  MPU6050

- Arduino IDE version (found in Arduino -> About Arduino menu):  NA - 2.0 or 1.8

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

When you use the ESP32 in Deep Sleep mode, and are using a MPU6050 that is configured to wake the ESP32, the interrupt status bits that are required to be read, will get demolished by the `init()` call (calling `reset()`), and the `begin()` call changes other settings that might not be desirable.

This is due to the fact that the ESP32's return from deep sleep, is very similar to a reset of the chip and many configurations, such as i2c connections, are reset and need to be reconfigured. It is important to be able to read the i2c, without disturbing important data.

An alternative could be to have `init()` not do anything but to make sure Wire and i2c are setup for the board so that communication can happen between the esp32 and the MPU6050. Then most people would use `begin()` to continue normal setup. Those who care to preserve the settings they already have, will not call `begin()`. They would read what they need, and change any settings as needed, without having a call that blanket resets things.

I don't think this is a big change, and I plan to try out those changes myself, to see if it accomplishes my goal. I'm curious what your thoughts are about this.

mahboud, 2022-03-07:
You can have a bootCount that can tell you whether it is the first time. Or you can check to see if any of the registers in the MPU6050 are setup the way you last set it up.

See the ESP32 example, "Deep Sleep with External Wake Up". There is a `RTC_DATA_ATTR int bootCount = 0;` that is only zeroed out when the device is reset, and not on deep sleep events.

I have a branch I'd like to PR, but I think I don't have enough permission? Do you have a preferred way to get PRs? Separate fork?

ladyada, 2022-03-07:
yes you must fork for a PR - we'd take a PR that does *not* break existing usage

mahboud, 2022-03-07:
That's my intention. To keep existing usage unchanged but add new functionality for those who need it.

mahboud, 2022-03-08:
One other way that this could be accomplished is if `begin()` had an optional argument, `doNotReset`. If it is called with that argument set to true, then it would check to make sure that some of the registers on the MPU had reasonable values, and would then refrain from resetting the MPU.

That's not how my PR is doing it, but it would remove the need for an additional method.