adafruit/Adafruit_MPU6050 #23: Examples do not build - code bug? - trying to return VOIDs

gitcnd, 2022-07-13:
Building for ESP32 D1 I observe the following:-

C```
:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp: In member function 'bool Adafruit_MPU6050::enableSleep(bool)':
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp:497:28: error: void value not ignored as it ought to be
   return sleep.write(enable);
                            ^
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp: In member function 'bool Adafruit_MPU6050::enableCycle(bool)':
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp:519:28: error: void value not ignored as it ought to be
   return cycle.write(enable);
                            ^
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp: In member function 'bool Adafruit_MPU6050::setGyroStandby(bool, bool, bool)':
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp:576:79: error: void value not ignored as it ought to be
   return gyro_stdby.write(xAxisStandby << 2 | yAxisStandby << 1 | zAxisStandby);
                                                                               ^
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp: In member function 'bool Adafruit_MPU6050::setAccelerometerStandby(bool, bool, bool)':
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp:603:40: error: void value not ignored as it ought to be
                            zAxisStandby);
                                        ^
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp: In member function 'bool Adafruit_MPU6050::setTemperatureStandby(bool)':
C:\Users\cnd\Documents\Arduino\libraries\Adafruit_MPU6050\Adafruit_MPU6050.cpp:622:28: error: void value not ignored as it ought to be
   return temp_stdby.write(1);
                            ^
exit status 1
Error compiling for board ESP32 Dev Module.

```

caternuson, 2022-07-13:
I just did a test compile of the `basic_readings` example for a target board of "ESP32 Dev Module" and it compiled without any errors.

Are all your libraries and board support packages up to date?

gitcnd, 2022-07-14:
Thanks for testing - I've been doing IoT for more than a decade, so it's a certainty that most of my other libs are out-of-date.

For anyone else facing this problem, I recommend the MPU6050 lib from ElectronicCats instead: https://github.com/ElectronicCats/mpu6050

It works immediately, without any dependency-hell, and clearly shows how to use the alternate I2C addresses of 0x69 when necessary (which is: usually).

closed 2022-07-14 by gitcnd