adafruit/Adafruit_MPU6050 #24: Can esp32 not be used?

bbhxwl, 2022-07-17:
[image]

[image]
[image]

asjdf, 2022-09-21:
Try to add `Wire.begin(21, 22);` before the `mpu.begin()`.
I think this problem is worth a PR.

caternuson, 2022-09-21:
The `basic_readings` example sketch runs fine here on a Feather ESP32.
```
MPU6050 Found!
Accelerometer range set to: +-8G
Gyro range set to: +- 500 deg/s
Filter bandwidth set to: 21 Hz

Acceleration X: 6.34, Y: -7.23, Z: -4.28 m/s^2
Rotation X: -0.05, Y: 0.01, Z: -0.01 rad/s
Temperature: 23.15 degC

Acceleration X: 6.33, Y: -7.23, Z: -4.28 m/s^2
Rotation X: -0.05, Y: 0.01, Z: -0.01 rad/s
Temperature: 23.27 degC

Acceleration X: 6.34, Y: -7.24, Z: -4.30 m/s^2
Rotation X: -0.05, Y: 0.01, Z: -0.01 rad/s
Temperature: 23.34 degC
```

Please post in the forums with photos of your setup:
https://forums.adafruit.com/

closed 2022-09-21 by caternuson