adafruit/Adafruit_MPU6050 #25: wrong documentation in V2.2.1 headerfile line27

GregorB54321, 2022-07-21:
original:
#define MPU6050_I2CADDR_DEFAULT                                                \
  0x68                         ///< MPU6050 default i2c address w/ AD0 high
#define MPU6050_DEVICE_ID 0x68 ///< The correct MPU6050_WHO_AM_I value

correct: "w/ AD0 **low**"

Reference: Product Specification Datasheet Rev.3.4 Page 33 paragraph 9.2
https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf

caternuson, 2022-07-21:
[image]

Yep, looks like. Want to submit a PR to update that?

caternuson, 2026-01-30:
closing. fixed with #36