adafruit/Adafruit_MPU6050 #27: The method setTemperatureStandby() always disables the sensor.

asaujim, 2022-10-15:
`setTemperatureStandby` always disables the temperature sensor regardless of the argument's value. The last line of its code (622) always writes a 1 instead of using the argument. So using `setTemperatureStandby(false)` still disables the sensor.

https://github.com/adafruit/Adafruit_MPU6050/blob/266337b2b97db655d328f557f5df2c76ea9c2306/Adafruit_MPU6050.cpp#L606-L623

It might just be a small typo. It seems to be solved by swapping the call in the line 622 for `return temp_stdby.write(enable);`

caternuson, 2022-10-15:
Agree it looks like a typo. And suggested fix looks proper. Want to PR that change?

asaujim, 2022-10-15:
Yes, I've already done it. It is the first PR that I do, please tell me if it needs any change. Thank you.

closed 2022-10-22 by asaujim