adafruit/Adafruit_MPU6050 #32: setMotionInterrupt is enabling the wrong bit

Nurgak, 2023-08-04:
- Arduino board:  ESP32
- Arduino IDE version: 1.8.19
- Library version: 2.2.4

I am trying to use the interrupt pin to trigger reads on the MCU when new data is ready, but I was not having any luck. I took a close look at how the library was enabling the pin interrupt and it did not make much sense when comparing to the datasheet.

This library has been around for a while and no issues have addressed this, so maybe I am misunderstanding something.

The interrupt is enabled by [`setMotionInterrupt()`](https://github.com/adafruit/Adafruit_MPU6050/blob/master/Adafruit_MPU6050.cpp#L416):
```c
void Adafruit_MPU6050::setMotionInterrupt(bool active) {
  Adafruit_BusIO_Register int_enable =
      Adafruit_BusIO_Register(i2c_dev, MPU6050_INT_ENABLE, 1);
  Adafruit_BusIO_RegisterBits int_motion =
      Adafruit_BusIO_RegisterBits(&int_enable, 1, 6);
  int_motion.write(active);
}
```
However, the [datasheet](https://invensense.tdk.com/wp-content/uploads/2015/02/MPU-6000-Register-Map1.pdf) shows that the relevant registry bits are the following:
[image]

As you can see, **the bit `6` does not have any function**.

I was interested in having the MPU6050 interrupt (`INT`) pin latched when new data was available. I changed the function to enable the bit `0` instead of `6` and indeed data started flowing as expected:
```c
void Adafruit_MPU6050::setMotionInterrupt(bool active) {
  Adafruit_BusIO_Register int_enable =
      Adafruit_BusIO_Register(i2c_dev, MPU6050_INT_ENABLE, 1);
  Adafruit_BusIO_RegisterBits int_motion =
      Adafruit_BusIO_RegisterBits(&int_enable, 1, 0);
  int_motion.write(active);
}
```

I noticed the same issue with the [`Adafruit_MPU6050::getMotionInterruptStatus()`](https://github.com/adafruit/Adafruit_MPU6050/blob/master/Adafruit_MPU6050.cpp#L430) method, which also reads the bit 6, which does not have any function in the datasheet.
[image]

This brings me to the [motion_detection.ino](https://github.com/adafruit/Adafruit_MPU6050/blob/master/examples/motion_detection/motion_detection.ino#L37) example: I have no idea how it actually works, as the bit 6 is being evaluated via the `getMotionInterruptStatus()` method and according to the datasheet nothing should happen. I can only guess this bit might _also_ be toggled, but is not documented as such.

Could someone double check this?

Relevant to this topic: the other thing I want to do is to enable `INT_RD_CLEAR` bit, in the `INT_PIN_CFG` register, so _any_ read would clear the latch, set by the available `setInterruptPinLatch()` method, but this is not exposed by this library. Currently, only reading the `INT_STATUS` register will clear the latch, which is not needed when you already know new data is available via the `INT` pin. Indeed, when I enabled this bit, executing `getMotionInterruptStatus()` became unnecessary.

saintaardvark, 2023-08-19:
Hi @Nurgak -- I'm not a developer for this library, but I came across your issue & wanted mention this page:

* https://www.eluke.nl/2016/08/11/how-to-enable-motion-detection-interrupt-on-mpu6050/

...which in turn points to:

* http://www.i2cdevlib.com/devices/mpu6050#registers

In a nutshell, it looks like the register map doesn't contain all the information about the MPU6050; reverse engineering appears to show that the 6th bit does indeed need to be set for motion detection to be enabled.

I'm only now just beginning a project to make use of this feature in the MPU6050.  Early tests with this driver and an ESP32 do seem to show this feature working.  My (rough, crappy) code looks like this:

```c++
            mpu->setMotionDetectionThreshold(1);
            mpu->setMotionDetectionDuration(1);
            mpu->setInterruptPinLatch(true);	// Keep it latched.  Will turn off when reinitialized.
            mpu->setInterruptPinPolarity(true);
            mpu->setMotionInterrupt(true);
```

With this, I can set the sensor on a table, give it a smack with my hand, and see an interrupt get generated.

I presume the developers of this library read the same resources and wrote the driver to use the features as described.

Hope that helps!

ladyada, 2023-08-20:
closing as this seems resolved?

closed 2023-08-20 by ladyada